si-machines/poli2 #29: Astra camera collision model is too large

Kukanani, 2018-01-26:
The model for the Astra is larger than the physical camera. I don't know the exact amount, but I would estimate that the collision model is about 25% larger than real size.

prashantrao, 2018-01-26:
The SolidWorks model is to scale as we have successfully designed and printed integration parts for the camera using the CAD model alone. The CAD model was provided by Orbbec. I believe the issue should be in the URDF unless that's what you already meant.

Kukanani, 2018-01-26:
Yes, I'm referring to the URDF. Possibly just the visual and not the collision mesh is incorrect, I have not checked the size of the collision mesh.

maxsvetlik, 2018-02-09:
This is because I used a URDF model and mesh that I found online. It seems to be for a different model of the Astra which was much larger. This also means that its visual frames would be off for any visual processing.

Kukanani, 2018-02-09:
I made a model of the Orbbec (not perfect but probably closer to actual size) last summer. Feel free to use if we can't get a model direct from Orbbec.

https://bitbucket.org/osrf/gazebo_models/src/4b4490dc9e938b42d2e9c8c7cce1bad177ad65f5/orbbec_astra_camera/?at=default

Kukanani, 2019-09-06:
No longer using Astra.

closed 2019-09-06 by Kukanani