si-machines/poli2 #30: Astra point cloud exhibits 2 different artifacts

Kukanani, 2018-01-30:
As discussed with @prashantrao recently, but putting it here to ensure that all requirements are captured.

There are currently two problems with the point clouds coming from the Astra:
 1. Points at the edges of objects are "smeared" towards and away from the camera. This is similar to the behavior seen with the Kinect2 driver, but in the case of the Kinect2 driver, there are alternative resolutions that work better. All point clouds from the Astra have the issue. This issue may be fixable by rebuilding the OpenNI2 or Astra drivers with additional options; see [this issue](https://github.com/orbbec/ros_astra_camera/issues/7).
 2. When objects get near the camera, the black area of the point cloud begins to be filled with complete nonsense values. The best way to see this is to create an Image display in RViz targeting /camera/depth/image_rect_raw or any other depth image, then hold your hand in front of the camera. These nonsense depth values result in lines of points radiating away from the camera that will throw off any attempts at quality perception. The [same GitHub issue referenced above](https://github.com/orbbec/ros_astra_camera/issues/7) may contain a fix for this bug as well, but it's unclear from reading the issue exactly what the fix addresses. This particular problem may persist even after building with the patch, in which case we should start looking at other camera options.

prashantrao, 2018-01-30:
https://github.com/mauriliodc/thin_astra

Don't know what "thin philosophy means" but is this helpful? Sorry, I am too far removed from this field to be of any help. However, I can google a lot better than assist meaningfully in this issue.

maxsvetlik, 2018-02-12:
Orbecc also offers the `Astra S` which is the close-ranged version of the Astra: (0.4 - 2m) vs (0.6 - 5m). If the issue is a harware limitation, this may not be a fix as the only difference is likely a focal length / lens change. But may be worth looking into nonetheless.

Kukanani, 2019-09-06:
No longer using Astra

closed 2019-09-06 by Kukanani