si-machines/poli2 #33: Update Documentation

maxsvetlik, 2018-02-09:
This can be open issue that people can voice areas that they feel documentation is lacking and/or incorrect.

Ideally once someone is aware that something is incorrect, they can fix it by making changes and committing to the `documentation` branch.

maxsvetlik, 2018-02-09:
- [ ] Remove details about old gripper
- [ ] Add details about new gripper

maxsvetlik, 2018-02-09:
- [ ] Arduino development / how to flash new code to arduinos / dependent libraries

maxsvetlik, 2018-02-12:
- [ ] Tilt motor launching script
- [ ] Wiring diagrams for each microcontroller
- [ ] Power block diagram
- [ ] Port names for each microcontroller

maxsvetlik, 2018-02-16:
[ ] - new charging scheme (removing two plugs from charging block, rather than removing serial charger from base)
[ ] - charging the aux battery
[ ] - plugging / unplugging / turning on aux battery

Kukanani, 2019-09-06:
The LaTeX documentation has been superseded and the issues listed are either fixed or no longer relevant.

closed 2019-09-06 by Kukanani