si-machines/poli2 #34: URDF does not reflect robot's gripper

maxsvetlik, 2018-02-09:
The URDF has the Weiss gripper as attached to the arm. This no longer reflects the actual robot state and should be changed to use the new gripper.

Additionally, the defacto gripper driver will no longer work as it is Weiss exclusive. A new driver will need to be found, added and integrated.

maxsvetlik, 2018-02-09:
MoveIt's SRDF will also need to be generated from the URDF. More specifically, convert from Xacro -> URDF and rerun setup_assistant

maxsvetlik, 2018-02-12:
The lack of a working driver causes issues for the TF tree. As a consequence, I've removed the arm from the URDF (commented out) and the `poli2_upstart` launch files. 

These should be added back and re-integrated once the gripper is added to the arm and the gripper driver is working.

Kukanani, 2019-09-06:
All issues here have been fixed except for one: the Robotiq does not update its state in RViz/MoveIt to reflect the actual gripper state.

Kukanani, 2019-09-09:
Fixed by e910221

closed 2019-09-09 by Kukanani