si-machines/poli2 #35: LED Control libraries not compatible with ROS

maxsvetlik, 2018-02-12:
We use two control libraries for changing LED state. 
- `Adafruit_DotStar` for the flexible led matrix array (eyes) 
- `APA102` for the flexible led strip (ears)

The `DotStar` library gives a number of useful functions that abstract bit manipulation away. However, it seems that this library is not compatible with `rosserial` due to what I'm assuming are synchronization issues. 

`APA102` does not have issues, however, so equivilent functions need to be written using the APA102 library that gives similar eye LED behavior. 

These can then be re-integrated as a rosservice through `rosserial`.

Until this happens, the eyes are not able to be controlled or turned on without a dedicated, non-rosified arduino.

maxsvetlik, 2018-02-13:
This turned out not to be a limitation of `DotStar` and did turn out to be a limitation of the Arduino Uno. 
Switching to a Mega and using its hardware SPI resolved the issue.

LED eyes + ears can be controlled via ros service calls.

maxsvetlik, 2018-02-13:
There are a couple Eye and Ear methods that do not work and need to be fixed. These are strictly aesthetic.

maxsvetlik, 2018-02-14:
Eye and Ear methods fixed- a working pulsing method (called breathing in-code) is working, as is a proper blinking capability that eliminates the need for a blocking `delay` call, which interrupts any animations on other LED strips.

prashantrao, 2018-02-14:
Wow look at you go.

PR #36 (2018-02-16): LED controller, limit switch
This introduces an LED controller driver for an Arduino Mega that also incorporates a limit switch used for initialization of the tilt motor encoders. The URDF of the pan/tilt system has been updated to match current physical design.

The driver / launcher of the tilt motor that utilizes this switch is in `poli_launch/scripts` and is called `launch_tilt_motor`. The specific behavior and the conditions in which the launcher succeeds is documented in the driver file. This closes #31.

The LED controller controls the Ear LED strip and Eye LED matrix through separate s [... cut]
closed 2018-02-16 by maxsvetlik