si-machines/poli2 #37: Kinetic MoveIt Upgrade Issues

vchu, 2018-02-15:
So it looks like MoveIt! Kinetic updated how it does status messages (away from just using booleans).

https://github.com/ros-planning/moveit/commit/4033e577b64ab7ef0e4473342168494f69ef9324

So as a result, if you install the latest kinetic moveit packages, the kinova_ros moveit demos break. My guess is that since we don't even use those files to remove them. It is also very quick to fix (just 2-3 lines of change), but I don't want to push it to the kinetic branch because it depends on a specific moveit package build. (I believe Poli2 is on a package prior to this change).

Putting this here to track, until we agree what to do with those files. The specific ones are:

> ~/catkin_ws/src/external/kinova_ros/kinova_moveit/kinova_arm_moveit_demo/src$`

	modified:   ../include/pick_place.h
	modified:   ../include/test_accuracy.h
	modified:   motion_plan.cpp

The error you'll see if you don't change these files are:

`catkin_ws/src/external/kinova_ros/kinova_moveit/kinova_arm_moveit_demo/src/motion_plan.cpp:177:11: error: cannot convert ‘moveit::planning_interface::MoveItErrorCode’ to ‘bool’ in assignment
   success = group.plan(my_plan);`

maxsvetlik, 2018-02-16:
Hm indeed. Since we're using the `kinetic_7dof` branch of the GT-RAIL fork of `kinova_ros` it may be worthwhile to open an issue there. 

I've been getting around this temporarily by adding a CATKIN_IGNORE file in `kinova_arm_moveit_demo`...

Kukanani, 2019-09-06:
This has been semi-permanently fixed by our setup scripts that automatically add a CATKIN_IGNORE in the moveit demo folder as Max mentioned above.

This could ultimately be fixed by rebasing our messy Kinova repo on the latest one from Kinova (which supports Kinetic), but that should be a separate issue.

re-open if it becomes an open issue again.

closed 2019-09-06 by Kukanani